chvmp/champ #2: 2D Nav Goal - robot moving in stationery position suring SLAM

bluetulip89, 2020-05-12:
[image]

While executing 2d nav goal  and giving goal to robot for SLAM, the 4 legs of the robot are moving but yet the robot is same position. 

Kindly look into it and help

bluetulip89, 2020-05-12:
The issued has been resolved by re-installing gazebo_ros_pkgs. I am using kinetic , so I did cloned the following and did catkin_make again. After this SLAM worked fine . 

Here is the link I used
git clone https://github.com/ros-simulation/gazebo_ros_pkgs.git -b kinetic-devel

closed 2020-05-12 by bluetulip89

grassjelly, 2020-05-12:
thanks for spotting this. fixed https://github.com/chvmp/champ/commit/35e6457ac0e3831560e9870fce43eaa0f40b5926