chvmp/champ #3: Running this on ROS-noetic

m2kulkarni, 2020-06-07:
I am trying to run this on latest ros version but it doesnt seem to work. When i follow the installation steps, catkin_make runs successfully but with error for yocs_velocity smoother. 
[image]
I also tried cloning yocs_velocity_smoother from upstream, but still it persists

bluetulip89, 2020-06-08:
This champ package is intended for kinetic or melodic. For the latest release we need to make some modifications to work. 

The "ecl_threads" ROS package is missing. Kindly install the package from this link (http://wiki.ros.org/ecl_threads) and let me know if you have any other errors.

m2kulkarni, 2020-06-08:
Hey, I did that, i built from source ecl packages and when i did catkin_make, i had no errors in that. It built successfully(no ecl errors like last time). But when i try to run `roslaunch champ_config bringup.launch rviz:=true` i get rviz with only a white mesh and lots of error in terminal
[image]

Also when i run `roslaunch champ_config gazebo.launch` gazebo gets stuck on preparing your world without any errors.

Maybe champ requires other packages which arent included? If so can you list them. Thanks. 
1. Hector_sensors_description

grassjelly, 2020-06-08:
hi, can you try this branch? I suspect it's publishing the transforms before the controller has started. Temporarily added some flag to ensure the controller has started before publishing data.

m2kulkarni, 2020-06-09:
Which Branch? Master or the noetic-bug branch?

grassjelly, 2020-06-09:
sorry missed the link. Updated my comment.

m2kulkarni, 2020-06-09:
Still no change, same thing happens :/

grassjelly, 2020-06-09:
did you rebuild the workspace and source it again?

m2kulkarni, 2020-06-09:
Yes, did that.

grassjelly, 2020-06-09:
Sorry can’t replicate this issue in Melodic. I suspect it could be related to the pose estimation that was recently added. Meanwhile you can give the previous commit a try. https://github.com/chvmp/champ/tree/5e0861d60e2e1b8597db93b3adbc5562f4bd31fd

m2kulkarni, 2020-06-09:
I tried that too. Here are the output files for 
1. git show-branch
2. catkin_make
3. roslaunch champ_config bringup.launch rviz:=true 
I did not find anything peculiar.
[out.txt](https://github.com/chvmp/champ/files/4753635/out.txt)

grassjelly, 2020-06-17:
hey @m2kulkarni . Saw your SLAM benchmarks. Did you manage to run this on Noetic?

lenia209911, 2020-07-22:
Hello
Can you please help?
Why doesn't gazebo.launch (mini_cheetah_config) get started, but just lies down on the ground? It feels like it won't launch TF
* Ubuntu 18.04 ros-melodic gazebo 9.13.1
[image]

grassjelly, 2020-07-22:
@lenia209911 hi, can you please post this as a separate issue.

nickle-fang, 2020-09-01:
@grassjelly I meet the same problem with @m2kulkarni . My OS is Ubuntu 20.04.

flynneva, 2020-09-23:
hey so i just built this package on the master branch on Ubuntu 20.04 using noetic. it builds fine as long as you have the dependencies installed.

the problem i think you are running into is one that `rosdep install` isnt actually installing your dependencies.  [I ran into the same problem and have created an issue](https://github.com/ros-infrastructure/rosdep/issues/779) over on the `rosdep` repo to track it.

to run `rosdep install` on noetic on ubuntu 20.04 run it while specifying your OS's version:
```
rosdep install --os=ubuntu:focal
```

flynneva, 2020-09-23:
what is a problem though is that it looks like some dependencies are not available yet for noetic:
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
yobotics_description: Cannot locate rosdep definition for [hector_sensors_description]
mini_ros: Cannot locate rosdep definition for [hector_sensors_description]
champ_navigation: Cannot locate rosdep definition for [hector_mapping]
opendog_description: Cannot locate rosdep definition for [hector_sensors_description]
dkitty_description: Cannot locate rosdep definition for [hector_sensors_description]
champ_description: Cannot locate rosdep definition for [hector_sensors_description]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully
```

flynneva, 2020-09-24:
created tracking issue for the Noetic dependencies #15

closed 2021-01-30 by grassjelly