chvmp/champ #5: Don't start gazebo.launch mini_cheetah_config

lenia209911, 2020-07-22:
Hello
Can you please help?
Why doesn't gazebo.launch (mini_cheetah_config) get started, but just lies down on the ground? It feels like it won't launch TF

Ubuntu 18.04 ros-melodic gazebo 9.13.1
[image]

grassjelly, 2020-07-22:
Hi,

The Mini Cheetah's URDF in the preconfigured robots is barebones. It doesn't include the inertia and transmissions required to be simulated in Gazebo. You can check out the default robot for reference https://github.com/chvmp/champ/blob/master/champ_description/urdf/champ.urdf

lenia209911, 2020-07-22:
The champ robot is also in the gazebo)

grassjelly, 2020-09-09:
You can follow the following commits how to integrate URDFs with Gazebo https://github.com/moribots/spot_mini_mini/pull/7/commits

grassjelly, 2020-09-09:
A working Gazebo model for the Mini Cheetah is now available. Please check  https://github.com/chvmp/robots  
[image]

closed 2020-09-09 by grassjelly