chvmp/champ #6: mini_cheetah cannot work in gazebo

Jonathan-Twz, 2020-07-22:
Hi, I've tried your demo for mini_cheetah, with command `roslaunch mini_cheetah_config bringup.launch` and then `roslaunch mini_cheetah_config gazebo.launch`. But the robot cannot stand up in gazebo, and come up with the information below:  

[FATAL] [1595421094.629746262]: Failed to unload nodelet '/velocity_smoother` from manager `nodelet_manager'
[WARN] [1595421124.845966, 1.244000]: Controller Spawner couldn't find the expected controller_manager ROS interface.

Could you please help me what's wrong with my command or how could I fix this bug.

grassjelly, 2020-07-22:
Hi there, this is the same issue as this https://github.com/chvmp/champ/issues/5 . You need to add the physics elements required for Mini Cheetah.  Meanwhile, you can run it in rviz without modifying the barebones URDF.

grassjelly, 2020-09-09:
A working Gazebo model for the Mini Cheetah is now available. Please check  https://github.com/chvmp/robots  
[image]

closed 2020-09-09 by grassjelly