chvmp/champ #7: [rospack] Error: package 'champ_arm_description' not found

Tomato1107, 2020-08-01:
When run the demo of the chicken_head
here shows some errors:

[ERROR] [1596271561.413624789]: Could not load resource [package://champ_arm_description/meshes/base.stl]: Unable to open file "package://champ_arm_description/meshes/base.stl".
[ WARN] [1596271561.414504831]: OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://champ_arm_description/meshes/base.stl in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
[ERROR] [1596271561.416309663]: Could not load model 'package://champ_arm_description/meshes/base.stl' for link 'base': OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://champ_arm_description/meshes/base.stl in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
[ WARN] [1596271561.416409552]: OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://champ_arm_description/meshes/base.stl in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
[ERROR] [1596271561.416494989]: Could not load model 'package://champ_arm_description/meshes/base.stl' for link 'base': OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://champ_arm_description/meshes/base.stl in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
[ WARN] [1596271561.416582220]: OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://champ_arm_description/meshes/base.stl in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
[ERROR] [1596271561.416653891]: Could not load model 'package://champ_arm_description/meshes/base.stl' for link 'base': OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://champ_arm_description/meshes/base.stl in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
[ WARN] [1596271561.416743030]: OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://champ_arm_description/meshes/base.stl in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
[ERROR] [1596271561.416816528]: Could not load model 'package://champ_arm_description/meshes/base.stl' for link 'base': OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://champ_arm_description/meshes/base.stl in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
[rospack] Error: package 'champ_arm_description' not found
[librospack]: error while executing command
[ERROR] [1596271561.440342792]: Could not load resource [package://champ_arm_description/meshes/lower_arm.stl]: Unable to open file "package://champ_arm_description/meshes/lower_arm.stl".
[ WARN] [1596271561.440626220]: OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://champ_arm_description/meshes/lower_arm.stl in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
[ERROR] [1596271561.440760767]: Could not load model 'package://champ_arm_description/meshes/lower_arm.stl' for link 'lower_arm': OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://champ_arm_description/meshes/lower_arm.stl in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
[ WARN] [1596271561.440883986]: OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://champ_arm_description/meshes/lower_arm.stl in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
[ERROR] [1596271561.440974053]: Could not load model 'package://champ_arm_description/meshes/lower_arm.stl' for link 'lower_arm': OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://champ_arm_description/meshes/lower_arm.stl in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
[ WARN] [1596271561.441084635]: OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://champ_arm_description/meshes/lower_arm.stl in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
[ERROR] [1596271561.441180609]: Could not load model 'package://champ_arm_description/meshes/lower_arm.stl' for link 'lower_arm': OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://champ_arm_description/meshes/lower_arm.stl in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
[ WARN] [1596271561.441287857]: OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://champ_arm_description/meshes/lower_arm.stl in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
[ERROR] [1596271561.441363586]: Could not load model 'package://champ_arm_description/meshes/lower_arm.stl' for link 'lower_arm': OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://champ_arm_description/meshes/lower_arm.stl in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
[rospack] Error: package 'champ_arm_description' not found
[librospack]: error while executing command
[ERROR] [1596271561.455210869]: Could not load resource [package://champ_arm_description/meshes/upper_arm.stl]: Unable to open file "package://champ_arm_description/meshes/upper_arm.stl".
[ WARN] [1596271561.455403346]: OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://champ_arm_description/meshes/upper_arm.stl in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
[ERROR] [1596271561.455519112]: Could not load model 'package://champ_arm_description/meshes/upper_arm.stl' for link 'upper_arm': OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://champ_arm_description/meshes/upper_arm.stl in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
[ WARN] [1596271561.455634574]: OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://champ_arm_description/meshes/upper_arm.stl in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
[ERROR] [1596271561.455724985]: Could not load model 'package://champ_arm_description/meshes/upper_arm.stl' for link 'upper_arm': OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://champ_arm_description/meshes/upper_arm.stl in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
[... cut ...] 9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
[ERROR] [1596271561.456095937]: Could not load model 'package://champ_arm_description/meshes/upper_arm.stl' for link 'upper_arm': OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://champ_arm_description/meshes/upper_arm.stl in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)

grassjelly, 2020-08-01:
hi,

Thanks for reporting the bug. I just fixed this with https://github.com/chvmp/champ/commit/f66a109780caca455ffd576ffafe1c1c3717addf

Please update your champ package.

Tomato1107, 2020-08-02:
Thanks very much!

closed 2020-08-02 by grassjelly