chvmp/champ #9: Champ, Minicheetah cannot be displayed in Gazebo

HitSZwang, 2020-08-19:
I use Ubuntu16.04 ROS KInetic gazebo8 and gazebo7 as well as attempts to operate in champ's gazebo environment.

The RVIZ environment displays and controls the movement of the robot correctly, and Gazebo opens the Gazebo file I created myself. The launch file generated with champ_setup_assistant cannot be opened, and champ_config Gazebo.launch is also available in your code。
[image]
thanks a lot

grassjelly, 2020-08-19:
May I know what's your exact Gazebo version? I remember having problems with Gazebo 7.0.0.Have you tried upgrading it? https://github.com/chvmp/champ_install/blob/master/install#L72-L75

Does the default demo works? ie. roslaunch champ_config gazebo.launch?

HitSZwang, 2020-08-19:
gazebo7  gazebo8.6  
the default demo  does not work

grassjelly, 2020-08-19:
Can you try removing this https://github.com/chvmp/champ/blob/master/champ_gazebo/launch/gazebo.launch#L12 or load your own world and see if it works

HitSZwang, 2020-08-20:
It works for my own world, but for **champ_config/gazebo.launch**,  **mini_cheetah_config/gazebo.launch** and **champ_gazebo/gazebo.launch** it does not works.

grassjelly, 2020-08-20:
Do you mind sharing your .world file?

HitSZwang, 2020-08-20:
my.launch
<launch>
    <arg name="paused" default="true"/>
    <arg name="use_sim_time" default="true"/>
    <arg name="gui" default="true"/>
    <arg name="headless" default="false"/>
    <arg name="debug" default="false"/>
    
    <arg name="user_debug" default="false"/>
     <arg name="description_file"           default="$(find yobotics_description)/xacro/robot_with_kinect.xacro"/>
     <arg name="ros_control_file" default="$(find yobotics_gazebo)/config/ros_control.yaml" />
    
    <include file="$(find gazebo_ros)/launch/empty_world.launch">
        <arg name="world_name" value="$(find yobotics_gazebo)/launch/world/normal.world"/> 
        <arg name="debug" value="$(arg debug)"/>
        <arg name="gui" value="$(arg gui)"/>
        <arg name="paused" value="$(arg paused)"/>
        <arg name="use_sim_time" value="$(arg use_sim_time)"/>
        <arg name="headless" value="$(arg headless)"/>
    </include>
 
    
      <param name="robot_description"
           command="$(find xacro)/xacro --inorder '$(arg description_file)' DEBUG:=$(arg user_debug)"/>
  
    
    <node pkg="gazebo_ros" type="spawn_model" name="urdf_spawner" respawn="false" output="screen"
          args="-urdf -z 0.5 -model yobotics_gazebo -param robot_description -unpause "/>

    
      <rosparam file="$(arg ros_control_file)" command="load"/>
        <node name="controller_spawner" pkg="controller_manager" type="spawner" respawn="false" output="screen" args="
            joint_states_controller
            joint_group_position_controller
            ">
        </node>

    
    <node pkg="robot_state_publisher" type="robot_state_publisher" name="robot_state_publisher"
          respawn="false" output="screen">
        <remap from="/joint_states" to="/yobotics_gazebo/joint_states"/>
    </node>

   

    

</launch>

HitSZwang, 2020-08-20:
thanks a lot. maybe I should wait for long time. it works now. But How to control the robot move in gazebo , the champ cannot move in gazebo.  it can move when I control uio jkl in RVIZ, but if I open the gazebo, the robot cannot move in RVIZ and gazebo. thank you !

grassjelly, 2020-08-20:
Can you post it as separate question? I'm closing this

closed 2020-08-20 by grassjelly

jash101, 2020-09-13:
Hello @HitSZwang you got it working just by waiting for "long time"? How long did you have to wait? Thanks

grassjelly, 2020-09-13:
Hey @jash101 also please make sure you're using the latest description package https://github.com/chvmp/mini-cheetah-gazebo-urdf . @HitSZwang has not merged my PR yet so you have to directly clone it from chvmp for now.

jash101, 2020-09-13:
Hey @grassjelly the README in your link says gazebo 8.6 is required, while I have gazebo 7.16 on my Ubuntu 16.04. Will this be an issue? I'd also like to mention that your suggested workaround of commenting out https://github.com/chvmp/champ/blob/master/champ_gazebo/launch/gazebo.launch#L12 works, although the robot slips a lot and mapping fails(I think due to no friction)

grassjelly, 2020-09-13:
@jash101 
1. It should work on 7.16. I was using the same version before I migrated to Melodic.
2. Sorry I didn't quite get what you mean about commenting out and which part of the README you are referring to. Could you open a new issue detailing what's wrong? Thank you!

HitSZwang, 2020-11-03:
hey, @grassjelly what did you do to make the mini_cheetah's real-time-factor = 1 i? thanks a lot.

grassjelly, 2020-11-03:
Hey @HitSZwang  take a look at this issue: https://github.com/chvmp/champ/issues/14