chvmp/champ #10: My slam.launch Can't display the map

HitSZwang, 2020-08-24:
/home/quadruped/Pictures/123.png
In my model_config, I can control the movement of the robots in gazebo and rviz by controlling champ_teleop, but the slam.launch of the mode_config exported by my self-check through the assistant cannot display the map, which is very dark. I can’t press 2D Nav Goal to control the movement of the robot, as shown in the picture. Thanks.

HitSZwang, 2020-08-24:
[image]

grassjelly, 2020-08-24:
You need to add sensors (LIDAR and IMU) in your robot's model for this demo to work. You can include this xacro file as a reference: https://github.com/chvmp/champ/blob/master/champ_description/urdf/champ.urdf.xacro#L43

grassjelly, 2020-08-25:
Hey @HitSZwang the current config for MIT cheetah hasn't been configured to work in Gazebo yet. Are you okay to include it in the supported robots? Thanks!

HitSZwang, 2020-08-26:
ok @grassjelly but i dont know how to include it. and my robot is imitative robot which is same to mini cheetah

grassjelly, 2020-08-26:
@HitSZwang You just have to push it to your github repository and I'll link to your page.

HitSZwang, 2020-08-26:
https://github.com/HitSZwang/mini-cheetah-gazebo-urdf

grassjelly, 2020-08-26:
thanks! Will send a PR once I’ve tested everything

grassjelly, 2020-09-08:
@HitSZwang Do you have a downsized version of the STL files? I managed to make it work in Gazebo but the Real Time Factor keeps dropping, probably due to the huge file size of the meshes. Would appreciate if you can upload that to so I can do further tests before sending the PR. Thanks!

grassjelly, 2020-09-09:
A working Gazebo model for the Mini Cheetah is now available. Please check  https://github.com/chvmp/robots  
[image]

closed 2020-09-09 by grassjelly