chvmp/champ #11: catkin_make error

waiyc, 2020-09-07:
Hi, this the error I encountered during catkin_make. 

```
c++: fatal error: no input files
compilation terminated.
/bin/sh: 1: -std=c++11: not found
champ/champ_gazebo/CMakeFiles/contact_sensor.dir/build.make:62: recipe for target 'champ/champ_gazebo/CMakeFiles/contact_sensor.dir/src/contact_sensor.cpp.o' failed
make[2]: *** [champ/champ_gazebo/CMakeFiles/contact_sensor.dir/src/contact_sensor.cpp.o] Error 127
CMakeFiles/Makefile2:1780: recipe for target 'champ/champ_gazebo/CMakeFiles/contact_sensor.dir/all' failed
make[1]: *** [champ/champ_gazebo/CMakeFiles/contact_sensor.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

```

grassjelly, 2020-09-07:
Does it work when you remove this https://github.com/chvmp/champ/blob/master/champ_gazebo/CMakeLists.txt#L5

`add_compile_options(-std=c++11)`

waiyc, 2020-09-08:
thanks for the reply. I have to comment out L5 and L39 in order to make it work.

grassjelly, 2020-09-08:
great! Closing this.

closed 2020-09-08 by grassjelly

grassjelly, 2020-09-08:
@waiyc btw, which ROS distro are you using?

waiyc, 2020-09-08:
Is kinetic