chvmp/champ #12: configure multi-robot system

YandongJi, 2020-09-18:
Hi:), 
currently I'm configuring a 4-robot system based on your code. I added 3 more namespaces named "robot_name2", "robot_name3", "robot_name4", however, I'm now facing this problem:  

[ERROR] [1600397229.850322115, 0.001000000]: SpawnModel: Failure - model name champ already exists.
[ERROR] [1600397229.859497225, 0.001000000]: SpawnModel: Failure - model name champ already exists.
[ERROR] [1600397229.867593171, 0.001000000]: SpawnModel: Failure - model name champ already exists.

it seems I need to change "champ.urdf.xacro", but how to change that? THX!

YandongJi, 2020-09-18:
Thx for the fix!
what is the large black thing on top of the robot?

grassjelly, 2020-09-18:
Hey, thanks for spotting the bug. You can follow these instructions how to spawn multiple robots: https://github.com/chvmp/champ#36-spawning-multiple-robots-in-gazebo

The black thing on top is a 2D LIDAR.

grassjelly, 2020-09-18:
I'm closing this. Feel free to open this again if you encounter related problems.

closed 2020-09-18 by grassjelly