chvmp/champ #14: controlling of mini cheetah

YandongJi, 2020-09-21:
Hi, it seems that the controlling of mini-cheetah needs some improvement. When I use **teleop** and type "i" to make it go straight, it's not struggling but not going in expectation. But if type "<" to make it go backwards, it performs better. Wondering whether it's the problem of pid parameter or other bugs. Thx!

grassjelly, 2020-09-21:
Hi,

You can fine-tune the PID constants here: https://github.com/chvmp/robots/blob/master/configs/mini_cheetah_config/config/ros_control/ros_control.yaml#L23-L34

YandongJi, 2020-09-21:
yeah, it may be the problem of foot friction. the feet keep slipping while going forward. are there any methods I can adjust that? I checked friction in "default.world" but there I can only adjust the plane friction instead of foot friction.

grassjelly, 2020-09-21:
Sure, you can tweak it from here: https://github.com/chvmp/mini-cheetah-gazebo-urdf/blob/master/yobo_model/yobotics_description/xacro/gazebo.xacro#L183-L190 (this is only for front-left foot)

YandongJi, 2020-09-24:
my robots are still slipping. Is your robot in gazebo running well? (use teleop to control) Many thx!

grassjelly, 2020-09-24:
Yes! But I only tested it with one robot instance. May I know what's your current Real Time Factor in Gazebo while running multiple robots? How many robots are you running?

YandongJi, 2020-09-24:
I'm running 4 robots and the Real Time Factor is 0.13. Seems a bit low. I changed the desired RTF in .world files but can't solve the problem. Also could you try whether you can make it going straight by only typing "i"?

grassjelly, 2020-09-24:
Yes, I can control the robot straight when pressing 'i' in teleop. I suspect your RTF is too low to get the physics right. Have you tried spawning just one robot?

YandongJi, 2020-09-24:
yeah, I just tried on champ. The RTF is 0.2-0.4.

YandongJi, 2020-09-24:
Can I know your RTF?

grassjelly, 2020-09-24:
RTF's ~ 1.0 .Here's how the robot behaves on walking test

[image]

YandongJi, 2020-09-28:
Thx, a higher performance computer solved the problem.

closed 2020-09-28 by YandongJi

peci1, 2021-03-23:
@grassjelly I think telling that low RTF should be the major problem crippling champ is not right. It's actually quite the opposite on most machines with 4+ cores. Gazebo can scarcely utilize all cores and gets slowed down in e.g. rendering or some other single-threaded tasks, so there are still some cores with free computational power that can be e.g. used by CHAMP. And the slower is the simulation, the more time has each control loop to finish! So actually lower RTF should mean the same or better performance of CHAMP. I did an experiment on my machine which usually achieves about 0.6 RTF. I set it manually to 0.1 and CHAMP got crippled again. It seems to me there has to be a component of CHAMP that wrongly uses simulation time. I'll try to investigate as this is a blocker for me.

peci1, 2021-03-23:
To prove my point: here are plots of one joint from `/joint_group_position_controller/command` that differ only in RTF. CHAMP was given exactly the same cmd_vel in both cases. The x and y axes in both plots have the same scale and the rqt_plot was run with `use_sim_time` set to true.

RTF 0.1

[image]

RTF 1.0 (or what my computer can do, about 0.6-0.8)

[image]

peci1, 2021-03-23:
But the rate at which the commands are published is the same (200 Hz).

peci1, 2021-03-23:
Got it: https://github.com/chvmp/libchamp/blob/cc884617153c52c0fe7e288d191acd2ac95f1d8b/leg_controller/phase_generator.h#L80 . I'll try to send a PR allowing proper use of simulation time.

peci1, 2021-03-23:
The PR fixing running using simulation time is here: https://github.com/chvmp/champ/pull/48 .