chvmp/champ #15: Noetic dependency tracking

flynneva, 2020-09-23:
creating this issue to track the status of some dependencies for this package that haven't been ported over to Noetic yet.

- [slam_gmapping](https://github.com/ros-perception/slam_gmapping)
  - dependent on [openslam_gmapping](https://github.com/ros-perception/openslam_gmapping)
  - both compile on noetic ubuntu 20.04, just release hasnt been made
- [hector_mapping](https://github.com/tu-darmstadt-ros-pkg/hector_slam/tree/noetic-devel)
  - noetic-devel branch available
  - compiles fine on the noetic-devel branch
- [hector_gazebo](https://github.com/tu-darmstadt-ros-pkg/hector_gazebo)
  - compiles

grassjelly, 2020-09-24:
Thanks @flynneva

grassjelly, 2021-01-30:
Closing this now as the following packages have Noetic releases already. Thanks @flynneva

closed 2021-01-30 by grassjelly