chvmp/champ #17: Installing Dependencies Error

abdksyed, 2020-10-13:
Hi, (Noob Here)
I am very new to ROS and Robotics in general. I was trying to clone and install all dependencies on my ubuntu 18.04.5(on virtual box), using following commands
```
sudo apt install -y python-rosdep
cd <your_ws>/src
git clone --recursive https://github.com/chvmp/champ
git clone https://github.com/chvmp/champ_teleop
cd ..
rosdep install --from-paths src --ignore-src -r -y
```
I created a folder and cloned, and inside my folder, trying to run
```
rosdep install --from-paths src --ignore-src -r -y
```
But it's giving me an error:
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies (ROS distro is not set. Make sure `ROS_DISTRO` environment
variable is set, or use `--rosdistro` option to specify the distro, e.g. `--rosdistro indigo`):
```
[image]
I even tried, 
```
rosdep install --from-paths src --ignore-src --rosdistro melodic -r -y
```

But it isn't working. Can you please help me out.

And sorry, if it is a very basic and stupid question.

Regards

grassjelly, 2020-10-13:
Hi,

Can you try the following?
`source /opt/ros/melodic/setup.bash `
`cd <your_catkin_ws>`
`source devel/setup.bash`

Then continue with rosdep ...

abdksyed, 2020-10-13:
Hi,
I tried running the source command,
But it's giving
```
bash: /opt/ros/melodic/setup.bash: No such file or directory
```

I also ran
```
sudo rosdep init
rosdep update
```

But same error

grassjelly, 2020-10-13:
Might be related to this: https://answers.ros.org/question/253480/bash-optroskineticsetupbash-no-such-file-or-directory/

abdksyed, 2020-10-13:
When installing, python-rosdep.
It's automatically giving
```
Note, selecting 'python-rosdep2' instead of 'python-rosdep'
```
Is this a problem?

grassjelly, 2020-10-13:
can you please post what's the output when you run this command:
`ls /opt/ros`

abdksyed, 2020-10-13:
```
ls: cannot access '/opt/ros': No such file or directory
```

grassjelly, 2020-10-13:
No worries. It happens. :) I'll close this for now. Feel free to open a new one if you have any more doubts.

closed 2020-10-13 by grassjelly