chvmp/champ #18: Error When running base drive in RVIZ

abdksyed, 2020-10-15:
Hi,
When I am running the command
```roslaunch champ_config bringup.launch rviz:=true```
I am getting this error, and the RVIZ GUI is opening up, but I am unable to perform any task like moving forward or turning etc using keyboard controls of (u,i,o,j......).
[image]

grassjelly, 2020-10-15:
Can you help me check the following:

1. Have you build the workspace:

`cd ~/<your_ws>/ && catkin_make
`

2. You have sourced the workspace before running the demos:

`source ~/<your_ws>/devel/setup.bash
`

3. You have yocs_velocity_smoother in your '_src_' folder

```
cd ~/<your_ws>/src
git clone https://github.com/chvmp/yocs_velocity_smoother
cd ..
catkin_make
```

abdksyed, 2020-10-15:
I tried reinstalling everything from scratch, and it's working fine now.
Thanks for the help anyways.

closed 2020-10-15 by abdksyed

trunc8, 2020-10-25:
I am facing a similar issue without any explicit error messages on the terminal.
[image]
I am unable to move the bot using teleop.launch.
I double-checked and followed the steps mentioned [earlier](https://github.com/chvmp/champ/issues/18#issuecomment-709172112) by grassjelly. Even re-installing wasn't of help.
After firing up teleop.launch, performing `rostopic echo /cmd_vel` displays nothing. Likewise for `rostopic echo \body_pose`.
Any suggestions would be greatly appreciated!

abdksyed, 2020-10-25:
After launching both terminals (teleop and rviz), press and HOLD the key in the controls terminal (U or I or O), the bot will move

trunc8, 2020-10-25:
It did! Thanks a lot!

Aniruddha7, 2020-10-25:
It seems like the warning in the screenshot are not error messages but
related to velocities that meant sensors are not getting proper input to
move

trunc8, 2020-10-25:
Haha right! I spent a long time going through the code imagining that those warnings caused the bot to not move. Can we display a more explicit message in the teleop.launch to keep the key pressed?
[image]
The last line gave me the impression that it's constantly publishing to the velocity topics.

Aniruddha7, 2020-10-25:
lol! Yes, you just need to follow those instructions. In the code there is
a docstring for this message and you can make changes locally by adding the
statement. But you can check with the author of the GitHub if you want to
publish it .

offjangir, 2021-03-07:
I am having a little problem @grassjelly   whenever I run any config ANYmal or Champ itself  rviz does not show any colors or the robot doesn't appear correctly I attached a screen shot Please help!

[image]

offjangir, 2021-03-07:
@trunc8  do you have any suggestions?

grassjelly, 2021-03-07:
hey @offjangir are there any errors when you run the launch files?

offjangir, 2021-03-07:
@grassjelly  no , not any  error just 2 warnings  also this is happening for all the configurations

offjangir, 2021-03-07:
[ WARN] [1615125291.797946407]: Warning: Some linear velocity entries in parameter imu0_config are listed true, but an sensor_msgs/Imu contains no information about linear velocities
[ WARN] [1615125291.802205050]: Warning: imu/data is listed as an input topic, but all its update variables are false

grassjelly, 2021-03-07:
@can you post your rqt_graph here?

offjangir, 2021-03-07:
here it is
[image]

offjangir, 2021-03-08:
@grassjelly  can you something that i am doing wrong here