chvmp/champ #20: jetson nano AND teensy 4.1 : which one to use ?  light mode or not ?

elpimous, 2020-10-15:
hello.
question is in the title, lol

I have both cards. I could use your light version to move my robot.
Is it more interesting to use the jetson nano to calculate moves ??

Thanks a lot for your nice package !!
Vincent

grassjelly, 2020-10-16:
Hey Vincent,

Yeah I agree with you. The Lite version is maintained only because of legacy issues. The project started with that version so yeah I kept it. However, I'll still choose the Jetson - Teensy combo with the locomotion controller running on the SBC and low level control(actuation, IMU data acquisition) on Teensy. This setup makes it more convenient to shift computers if you need more computational power.

grassjelly, 2020-10-16:
hey @elpimous I'll close this for now. Feel free to open another issue if you have any more doubts.

closed 2020-10-16 by grassjelly