chvmp/champ #21: Working on open Quadrupeder (teensy 4.0) How-to?!

elpimous, 2020-10-16:
Hi.
Champ simulations 👍

Now real-time. 
Working on Jetson, connected to Adham/Maurice/Mano board, containing a teensy 4.0.

Teensy is rosserial ready. 

Do I just need to run the next lines, to be able to use your package on my Dog?

```
<!-- Sensor Interface Node -->
<node pkg="rosserial_python" type="serial_node.py" name="serial_node">
    <param name="port" value="/dev/ttyS0"/> <!-- HW Serial -->
    <param name="baud" value="500000"/>  <!-- must match Teensy -->
  </node>
```
Thanks a lot. 

Vincent.

grassjelly, 2020-10-24:
HI Vincent,

I’m not sure on Open Quadruped’s firmware but these guidelines might help you setup your real robot: https://github.com/chvmp/champ/wiki/Hardware-Integration.

What I would recommend is to write direct comms interface between your Teensy and Linux Machine.

closed 2020-11-04 by grassjelly

Quothd, 2021-03-18:
On the Jetson, you need to specify the serial port as: 

<param name="port" value="/dev/ttyTHS1"/>