chvmp/champ #27: Bot keeps collapsing

bs2603, 2021-01-26:


bs2603, 2021-01-26:
After I run the command-
roslaunch champ_config gazebo.launch 
 
Gazebo opens and the simulated bot collapses to the ground, I don't want to switch off gravity, what should I do?

grassjelly, 2021-01-26:
@bs2603 are you using your own .world file?

bs2603, 2021-01-26:
Nope,
here are the steps I am following-
- Open a terminal and the catkin workspace
- source the devel/setup.bash file
- Run the base driver using `roslaunch champ_config bringup.launch rviz:=true`
- Run the gazebo world using `roslaunch champ_config gazebo.launch `

I am using ROS melodic on ubuntu 18.04.

grassjelly, 2021-01-26:
can you post a gif or a video?

Aniruddha7, 2021-01-26:
I used to have similar issues but once you immediately terminate Gazebo and
relaunch RVIZ and Gazebo, the issue got fixed. Sometimes it also happens
due to RAM/PC constraints. Make sure no background process is running
behind the application.

bs2603, 2021-01-29:
I was able to resolve the issue,
This was the error-

_Could not load controller 'joint_group_position_controller' because controller type 'effort_controllers/JointTrajectoryController' does not exist._

I installed the controllers using -
sudo apt-get install ros-melodic-ros-control ros-melodic-ros-controllers

these controllers are not a part of the default melodic full desktop installation,thus need to be added separately.
Thanks for the help!

closed 2021-01-29 by bs2603