chvmp/champ #32: Unitree A1 robot fails to move in Gazebo and falls over

tahir1069, 2021-02-22:
I am trying to use the Gazebo simulation of A1 robot in Gazebo(the one in Champ package) but the robot fall over(I have added the desired properties for Gazebo in URDF). 

The gait planning of the robot is working fine [here](https://my.hidrive.com/lnk/vxCVyDcT) the robot can be seen being teleoperated in RVIZ. But when the same robot is taken to [Gazebo](https://my.hidrive.com/lnk/QTC1S9Jp) it is not able to teleoperate and falls after some time. To me there can be few reasons setting proper physics parameters, setting PID values for `ros_control` or may be something else which at the moment I don't know. 

Anyone who can guide/point out some problems with the simulation.

grassjelly, 2021-02-22:
Hey @tahir1069 this might be related to this issue https://github.com/chvmp/champ/issues/14

grassjelly, 2021-03-05:
Hey @tahir1069 . Did you manage to resolve this?

tahir1069, 2021-03-05:
Hi @grassjelly I was able to solve the problem.

closed 2021-03-05 by tahir1069

grassjelly, 2021-03-05:
hey @tahir1069 could you please share how you resolved it for other users' reference? Thanks!

tahir1069, 2021-03-06:
Well I solved the issue by changing the values for the PID and exactly following the steps in above mentioned issue. The default one generated by champ also work. One issue for me was the laptop power, by changing to higher performance machine I was able to solve the problem. Because on a lower performance machine feedback and physics calculation are hardly synchronized so a lack in this was also the problem. The package for A1 can be found [here](https://github.com/MYBOTSHOP/mbs_unitree_ros/tree/master/mbs_simulations/mbs_a1_gazebo).

Supergithubber, 2021-12-12:
> Well I solved the issue by changing the values for the PID and exactly following the steps in above mentioned issue. The default one generated by champ also work. One issue for me was the laptop [...]

Hi！
I have also encountered the same problem, but the url you sent is not accessible, can you upload it to GIhub again?
Thanks！