chvmp/champ #33: Build Issue related to libbullet version

dlee640, 2021-02-26:
Hello. I am getting a catkin_make error that is very weird.

I have Ubuntu18.04 with Gazebo9. I am hetting No package 'bullet' found errror, that seems unsolvable even by executing `sudo apt-get installl libbullet-dev`. Currently, my system says that I have libbullet 2.87 installed so I am very lost on why this is happening.

[image]
Has anyone had this issue or know how to solve it? I feel like this is a path issue related to gazebo..

grassjelly, 2021-02-27:
looks like you have a virtual environment activated. Could it be clashing with that?

grassjelly, 2021-03-05:
Hey @dlee640 . Did you manage to resolve this?

dlee640, 2021-03-05:
Yes, sorry for the late response.

closed 2021-03-05 by dlee640

grassjelly, 2021-03-05:
@dlee640 do you mind sharing how you resolve this for future references?

dlee640, 2021-03-05:
probably not the best solution, I reinstalled ubuntu.