chvmp/champ #34: Contact Sensor failing in gazebo.launch

ericnieves, 2021-03-03:
Hey all,

Whenever I try to run:

roslaunch champ_config gazebo.launch, I keep getting this error message:

**[contact_sensor-13] process has died [pid 7140, exit code -6, cmd /home/user/champ_ws/devel/lib/champ_gazebo/contact_sensor __name:=contact_sensor __log:=/home/user/.ros/log/bb818682-7c41-11eb-b59d-4485003e4a6d/contact_sensor-13.log].
log file: /home/user/.ros/log/bb818682-7c41-11eb-b59d-4485003e4a6d/contact_sensor-13*.log**

I deleted the built contact sensor out of the directory:

**/home/user/champ_ws/devel/lib/champ_gazebo**

And then ran **catkin_make** in the catkin workspace. Although the build was successful, including replacing the contact sensor, I got some warning messages that I think are related:

**/usr/bin/ld: warning: libboost_thread.so.1.65.1, needed by /usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libgazebo_transport.so, may conflict with libboost_thread.so.1.71.0**

**/usr/bin/ld: warning: libboost_filesystem.so.1.65.1, needed by /usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libgazebo_common.so, may conflict with libboost_filesystem.so.1.71.0**

**[100%] Built target contact_sensor**

I'm rather new to this, but I feel like there is an issue with conflicting boost libraries and I can't seem to figure out what to do. Can anyone advise me?

grassjelly, 2021-03-05:
I think this issue is similar to this https://github.com/ros-planning/moveit/issues/2090#issuecomment-629052057 . Closing this as the problem's not related to the repo

closed 2021-03-05 by grassjelly

offjangir, 2021-03-09:
@ericnieves  were you able to run it finally . out of nowhere i have same problem it was working fine sometime ago