chvmp/champ #36: champ bringup.launch imu/data rostopic not showing any messages

dlee640, 2021-03-05:
Hello.

When I launch champ bringup,
```
dlee640@dlee640-T580:~$ roslaunch champ_config bringup.launch rviz:=true
... logging to /home/dlee640/.ros/log/2feaeaac-7dcf-11eb-a764-b46bfc2dbb2b/roslaunch-dlee640-T580-23563.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://dlee640-T580:36939/

SUMMARY
========

PARAMETERS
 * /base_to_footprint_ekf/base_link_frame: base_link
 * /base_to_footprint_ekf/frequency: 50
 * /base_to_footprint_ekf/imu0: imu/data
 * /base_to_footprint_ekf/imu0_config: [False, False, Fa...
 * /base_to_footprint_ekf/odom_frame: base_footprint
 * /base_to_footprint_ekf/pose0: base_to_footprint...
 * /base_to_footprint_ekf/pose0_config: [True, True, True...
 * /base_to_footprint_ekf/transform_time_offset: 0.045
 * /base_to_footprint_ekf/transform_timeout: 0.01
 * /base_to_footprint_ekf/two_d_mode: False
 * /base_to_footprint_ekf/world_frame: base_footprint
 * /champ_controller/gazebo: False
 * /champ_controller/joint_controller_topic: joint_group_posit...
 * /champ_controller/publish_foot_contacts: True
 * /champ_controller/publish_joint_control: True
 * /champ_controller/publish_joint_states: True
 * /footprint_to_odom_ekf/base_link_frame: base_footprint
 * /footprint_to_odom_ekf/frequency: 50
 * /footprint_to_odom_ekf/imu0: imu/data
 * /footprint_to_odom_ekf/imu0_config: [False, False, Fa...
 * /footprint_to_odom_ekf/odom0: odom/raw
 * /footprint_to_odom_ekf/odom0_config: [False, False, Fa...
 * /footprint_to_odom_ekf/odom_frame: odom
 * /footprint_to_odom_ekf/transform_timeout: 0.01
 * /footprint_to_odom_ekf/two_d_mode: True
 * /footprint_to_odom_ekf/world_frame: odom
 * /gait/com_x_translation: 0.0
 * /gait/knee_orientation: >>
 * /gait/max_angular_velocity_z: 1.0
 * /gait/max_linear_velocity_x: 0.5
 * /gait/max_linear_velocity_y: 0.25
 * /gait/nominal_height: 0.2
 * /gait/odom_scaler: 0.9
 * /gait/pantograph_leg: False
 * /gait/stance_depth: 0.0
 * /gait/stance_duration: 0.25
 * /gait/swing_height: 0.04
 * /joints_map/left_front: ['lf_hip_joint', ...
 * /joints_map/left_hind: ['lh_hip_joint', ...
 * /joints_map/right_front: ['rf_hip_joint', ...
 * /joints_map/right_hind: ['rh_hip_joint', ...
 * /links_map/base: base_link
 * /links_map/left_front: ['lf_hip_link', '...
 * /links_map/left_hind: ['lh_hip_link', '...
 * /links_map/right_front: ['rf_hip_link', '...
 * /links_map/right_hind: ['rh_hip_link', '...
 * /robot_description: <?xml version="1....
 * /robot_state_publisher/ignore_timestamp: True
 * /robot_state_publisher/publish_frequency: 200
 * /robot_state_publisher/use_tf_static: False
 * /rosdistro: melodic
 * /rosversion: 1.14.10
 * /state_estimator/close_loop_odom: False
 * /tf_prefix: /
 * /velocity_smoother/accel_lim_v: 1.0
 * /velocity_smoother/accel_lim_w: 2.0
 * /velocity_smoother/decel_factor: 1.0
 * /velocity_smoother/frequency: 10.0
 * /velocity_smoother/robot_feedback: 2
 * /velocity_smoother/speed_lim_v_x: 0.5
 * /velocity_smoother/speed_lim_v_y: 0.35
 * /velocity_smoother/speed_lim_w: 0.4

NODES
  /
    base_to_footprint_ekf (robot_localization/ekf_localization_node)
    champ_controller (champ_base/quadruped_controller_node)
    footprint_to_odom_ekf (robot_localization/ekf_localization_node)
    nodelet_manager (nodelet/nodelet)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz (rviz/rviz)
    state_estimator (champ_base/state_estimation_node)
    velocity_smoother (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [23665]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 2feaeaac-7dcf-11eb-a764-b46bfc2dbb2b
process[rosout-1]: started with pid [23676]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [23683]
process[nodelet_manager-3]: started with pid [23684]
process[velocity_smoother-4]: started with pid [23685]
process[champ_controller-5]: started with pid [23691]
process[state_estimator-6]: started with pid [23698]
process[base_to_footprint_ekf-7]: started with pid [23699]
process[footprint_to_odom_ekf-8]: started with pid [23705]
process[rviz-9]: started with pid [23718]
[ INFO] [1614961439.086236643]: Successfully parsed urdf file
[ INFO] [1614961439.095383893]: Successfully parsed urdf file
**[ WARN] [1614961439.170921801]: Warning: Some linear velocity entries in parameter imu0_config are listed true, but an sensor_msgs/Imu contains no information about linear velocities
[ WARN] [1614961439.175079859]: Warning: imu/data is listed as an input topic, but all its update variables are false**
[ INFO] [1614961439.196014083]: rviz version 1.13.15
[ INFO] [1614961439.196089498]: compiled against Qt version 5.9.5
[ INFO] [1614961439.196106039]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1614961439.202383253]: Forcing OpenGl version 0.
[ INFO] [1614961439.747785297]: Stereo is NOT SUPPORTED
[ INFO] [1614961439.747863805]: OpenGl version: 3.0 (GLSL 1.3).

```
**I get those warnings above about how imu/data's updated variables are false, which I think explains why I am not getting any data for my /imu/data topic.**

So, I took a loot at yaml files within `/src/champ/champ_base/config/ekf` folder where some configurations are defined.

First, what does the topic `/base_to_footprint_pose` mean? Does it mean robot base's location? I am kind of confused on the terminology and would appreciate your explanation!

Also, I feel like there is a small bug in the config setting for imu0_config. 

The `imu0_config` in `base_to_footprint.yaml` specifies:
```
frequency: 50

transform_timeout: 0.01
transform_time_offset: 0.045

two_d_mode: false
# diagnostics_agg: true

#x     , y     , z,
#roll  , pitch , yaw,
#vx    , vy    , vz,
#vroll , vpitch, vyaw,
#ax    , ay    , az
...

imu0: imu/data
imu0_config: [false, false, false,
              false, false, false,
              true, true, true,
              false, false, false,
              false, false, false]
```
The `imu0_config` in `footprint_to_odom.yaml` specifies:
```
frequency: 50

transform_timeout: 0.01
# transform_time_offset: 0.05

two_d_mode: true
# diagnostics_agg: true

#x     , y     , z,
#roll  , pitch , yaw,
#vx    , vy    , vz,
#vroll , vpitch, vyaw,
#ax    , ay    , az

...

imu0: imu/data
imu0_config: [false, false, false,
              false, false, true,
              false, false, false,
              false, false,  true,
              false, false, false]
```

I'd really appreciate you taking a look at this small bug and providing me with some explanation! Thanks again for providing us with great resource.

grassjelly, 2021-03-08:
Hi the reason that those warnings are showing is because when you’re walking the robot without the physical robot/Gazebo (solely in RVIZ), there’s no IMU data sent to the Kalman Filter.

For the config matrices, imu0_config have different boolean values because each filter lies on different frames and uses fuses different velocity values from the IMU. You can find out more here: http://docs.ros.org/en/melodic/api/robot_localization/html/configuring_robot_localization.html

peci1, 2021-03-23:
Isn't the problem in the fact that base_to_footprint specifies linear velocities from IMU, which are not a part of the message? There are only linear accelerations and angular velocities there...

closed 2021-03-26 by dlee640