chvmp/champ #38: Bringup is failing to launch to the robot model in rviz

offjangir, 2021-03-08:
[image]
[image]

grassjelly, 2021-03-08:
hey @offjangir could you post your TF graph here too? Thanks

offjangir, 2021-03-08:
@grassjelly  the tf graph is completely empty
[image]

$rosrun tf view_frames
$evince frames.pdf

grassjelly, 2021-03-08:
strange. Do you have robot_state_publisher, robot_state_publisher, and robot_localization?

offjangir, 2021-03-08:
yes i have that. idk what the problem is.. gazebo simulations are running perfectly fine

grassjelly, 2021-03-08:
I can’t replicate this error. What’s your tf graph like when running in Gazebo?

offjangir, 2021-03-09:
no tf data @grassjelly 
plus again the launch is stuck at the starting screen for gazebo too 

[image]

offjangir, 2021-03-09:
even if it starts the champ/ANYmal doesnt load

offjangir, 2021-03-09:
[image]

offjangir, 2021-03-09:
[image]

offjangir, 2021-03-10:
[frames.pdf](https://github.com/chvmp/champ/files/6112232/frames.pdf)

offjangir, 2021-03-10:
i did have such problem in turtlebot3 i resolved it by export TURTLEBOT3_MODEL= modelname   
but I dont think it will work here

grassjelly, 2021-03-13:
Are you able to receive the joint_states ? Are you passing any argument to the launch file when you run bringup.launch?

grassjelly, 2021-03-19:
Hey @offjangir . Have you resolved this issue?

deardost, 2021-03-23:
Hi there @offjangir 
Just change the **Fixed Frame** under **Global Options** to `base_link`. It will resolve all the errors. Then change it back to `odom`
I had the same issue as you. I resolved it this way.

deardost, 2021-03-23:
Can anyone say the reason why it works only when you change it to `base_link` first before fixing it back to `odom` ?

grassjelly, 2021-03-23:
@deardos Does this happen when you run bringup.launch without passing rviz argument and launching the rviz config file independently? My suspicion is that odom->base_link is not published yet when RVIZ starts to load the asset.

deardost, 2021-03-23:
@grassjelly No. 
[image]
I follow the steps above exactly as they are. But even then, it happens that the robot is white and I can't perform the manual walking i-e teleop operations on the bot. When I change fixed frame to base_link, the robot gets everything corrected. Then I change it back to odom to make the robot, using keyboard.

deardost, 2021-03-23:
I am using ROS Noetic.

closed 2021-04-01 by offjangir