chvmp/champ #40: Autonomous navigation and obstacle avoidance

devrostech, 2021-03-17:
I spawned the CHAMP in willowgarage.world and wanted to perform SLAM. I used the following commands:

**$ roslaunch champ_config gazebo.launch gazebo_world:=/usr/share/gazebo-9/worlds/willowgarage.world
$ roslaunch champ_config slam.launch rviz:=true**

I started giving goal locations to the champ(initially very close goals) to start developing the map in RViz. It started to move towards the first goal and then when I tried to give another location, the robot got stuck in the door. 

Below is the screen capture:

[image]

I have also tried SPOT and other robots for this but they are also not performing well. 

Following are my questions: 
1. Is there some way I can avoid these obstacles and perform smooth mapping?
2. I even tried to use the outdoor.world simulation and the default map which has been already created and same issues are there that the robot is usually not avoiding the obstacles and also RViz does not show the robot in correct position and the robot loses the track of it's location. Also the map objects changes and it becomes a complete mess inside the map when the robot moves towards a different location
3. The robot usually does not completes it's goal target or moves in different direction or behaves weirdly when any goal location has been assigned.

Please help me!

Thank you

grassjelly, 2021-03-17:
Hi, your Real Time Factor’s (0.07 on the first photo) way too low which probably explains the weird behaviour.  Check out this thread https://github.com/chvmp/champ/issues/14

devrostech, 2021-03-17:
Thank you for the response. Can you try the same and post a video please? I tried it on RTF 0.4 and it was still the same. I saw you were getting RTF of 1.0 so it would be of great help.

devrostech, 2021-03-29:
I upgraded my machine and now I am getting an RTF of 1.0 and even then autonomous navigation and obstacle avoidance is having problems

grassjelly, 2021-06-10:
hey @devrostech Have you resolved the problem?

closed 2022-07-02 by grassjelly