chvmp/champ #42: bad performance in SLAM

mOZz339, 2021-03-19:
I used original parameters of gmapping in SLAM, the mini-cheetah got a bad performance, as following.

https://user-images.githubusercontent.com/52522771/111759487-9c358a00-88d8-11eb-939e-a574c1ce16ee.mp4

And ,this is my world.
[image]

grassjelly, 2021-03-19:
Hi can you try increasing the max range https://github.com/chvmp/champ/blob/master/champ_config/launch/include/gmapping.launch#L10

mOZz339, 2021-03-22:
> Hi can you try increasing the max range https://github.com/chvmp/champ/blob/master/champ_config/launch/include/gmapping.launch#L10

Thanks,it performs better than before,but I still can not solve the probem of instantaneous translation in the video (time is 0:31,1:32). It appears a lot during my SLAM,especially on the turning left or right. The reason is my world is too large or the shake of robot model?

grassjelly, 2021-03-22:
Looks like the robot's mapping on a featureless area of the environment. It's a known issue for mapping as discussed here https://answers.ros.org/question/12427/gmapping-failed-in-long-corridor-using-navigation-stack/ .

mOZz339, 2021-03-22:
> Looks like the robot's mapping on a featureless area of the environment. It's a known issue for mapping as discussed here https://answers.ros.org/question/12427/gmapping-failed-in-long-corridor-using-navigation-stack/ .

Thanks for replying,I will close the issue ^_^

closed 2021-03-22 by mOZz339