chvmp/champ #43: Feedback in the controller?

peci1, 2021-03-21:
Hi, I've tried CHAMP and noticed an unexpected thing. There seems to be no feedback going to the `champ_controller` node (looking at rqt_graph of the running `roslaunch champ_bringup gazebo.launch`). Is that right? And is there a way to plug feedback in the controller, so that it can e.g. keep correct body pose even when on an inclined plane? I know that complex terrain traversal needs something very different, but at least simple terrain could probably be handled by some kind of orientation feedback (and leg pose feedback).

Anyways, thanks a lot for the package. I'm now trying to make it work with Spot in Ignition Gazebo for the DARPA SubT competition. If I succeed, I'll let you know.

grassjelly, 2021-03-22:
Hey @peci1, the current implementation of the default controller is open-loop. It's assumed that the joint position controllers will get the actuators to the target joints generated by the controller but there's no mechanism to auto-correct the quad's body pose for now. One thing you can try is to implement a PID controller before it goes to the[ body controller](https://github.com/chvmp/champ/blob/master/champ_base/src/quadruped_controller.cpp#L93) and use the output from state estimation node/IMU as a feedback.

peci1, 2021-03-23:
Okay, thank you for the information. Maybe this should be mentioned in the readme? I think feedback is so common in robotics that it is really unexpected this controller works even in an open-loop setting (or better called a hybrid one with feedback in the individual motor PIDs).

closed 2021-03-23 by peci1