chvmp/champ #44: Spot in Ignition Gazebo

peci1, 2021-03-22:
I got Spot basically walking in Ignition Gazebo for SubT challenge (I had to reimplement gazebo_ros_control for Ignition Gazebo on the way, but now I'm looking for a better solution as Ignition Gazebo cooperation with ROS is not as easy as with Gazebo Classic).

I've got a few questions regarding CHAMP performance.

I had to change the PID parameters since SubT runs with a time-step of 4 ms instead of 1 ms. I just divided P, I and D by 4 and it worked. I then tuned the P constant to ~200 (original is 600, so after the division, it was 150). I also moved `com_x_translation` 5 more cm to the back, otherwise the robot walked much better backwards than forward.

However, it seems to me the motion is still a bit sloppy. Could you please comment on the performance? Is that what I should expect, or should I tune some more parameters to get it even better? Especially going backwards seems to suffer from a lot of slippage. Also, trying to go faster than approx. 0.7 m/s quickly leads to flipovers (mostly by losing side-to-side balance; is that related to #43?). But the spot model in CHVMP specifies max speed 1.5. Did you actually get it reliably working with this speed?

https://user-images.githubusercontent.com/182533/111930455-4d931600-8ab9-11eb-8b1e-f3e96fe01019.mp4

I tried playing with `mu1`, `mu2` and `kp` of the lower leg collisions, but it did not seem to have much effect.

I've also noticed you set `<maxVel>` in the URDF `<gazebo>` tag to 0. Why is that? Shouldn't the contact correcting velocity be left there? Or does 0 mean that any velocity is allowed?

And which parameter affects the height of the jump you can see in the end of the video? (that's what happens if I press down arrow on D-pad and release it).

grassjelly, 2021-03-22:
Hey @peci1 this is the best fine-tuned [parameters](https://github.com/chvmp/robots/blob/master/configs/spot_config/config/ros_control/ros_control.yaml#L23-L34) I've tried for Spot so far . I believe this can be tuned more as I got quadrupeds similar to Spot's footprint walking more reliably (AnymalB and AnymalC). As a reference here's Spot walking using the parameters I shared 
[image] . 
[image]

I haven't tried running the robot at 1.5 m/s but I think it's possible with a lower [stance_duration](https://github.com/chvmp/robots/blob/master/configs/spot_config/config/gait/gait.yaml#L10) value (~ 0.1 - 0.3)

Can't really remember why I set the maxVel to zero (could be a typo) although I just tried setting it to 1.0 seems to make no difference at all.

I recommend to work on the default gait parameters and fine-tune the PID found in the ros_control folder.

peci1, 2021-03-23:
Thank you for the ideas. I'm gonna try to tune them a bit more.

peci1, 2021-04-15:
It is there! https://app.ignitionrobotics.org/OpenRobotics/fuel/models/bosdyn_spot

Connect with CHAMP using support package https://github.com/ctu-vras/subt_spot . To be used successfully, it needs a few support plugins, too. You can find them in https://github.com/ctu-vras/subt/tree/spot/submitted_models/bosdyn_spot/launch in files common.rb and vehicle_topics.launch.

closed 2021-04-15 by peci1

grassjelly, 2021-04-15:
@peci1 awesome man! Thanks!

jediofgever, 2022-05-20:
hi @peci1, @grassjelly ,

I am running the spot on ROS2 with ported champ [code](https://github.com/jediofgever/quadruped_ros2). I used the same configuration for spot as with @peci1's. 
I am facing some issues as robot flips after some steps. I haven't figured out the reason for this, I have tried some other quadrupeds as well and still facing the similar problems. Is anyone aware of a quadruped successfully locomating in ROS2 in Gazebo?  

Here are some videos that I captured from different quads that I have tried; 

Spot with same configuration posted in [peci1's repo](https://github.com/ctu-vras/subt/tree/spot/submitted_models/bosdyn_spot);
Running on [champ controller](https://github.com/jediofgever/quadruped_ros2);

https://user-images.githubusercontent.com/18460336/169563585-82831d4a-2346-4779-b697-2cd910461a87.mp4

The default champ quadruped ;
Running on [champ controller](https://github.com/jediofgever/quadruped_ros2);

https://user-images.githubusercontent.com/18460336/169563765-016990e1-a81f-4290-b5eb-bec7a0a08bf9.mp4

Yet another quadruped but it has no grip of the ground, it just walks in its place;
Running on a [different](https://github.com/jediofgever/notspot_sim_cpp);

https://user-images.githubusercontent.com/18460336/169563691-02e6941d-2545-4dea-ac64-bea567009658.mp4

All above robots are running on ROS2, I have not tried them on ROS1 personally but based on videos I have seen, they do behave correctly, there must be change that I can't get my head around. with same `urdf`, `ros_control` and other parameters the behaviour is different. I would appreciate if you have any inputs regarding this. 

Best.

peci1, 2022-05-20:
Hi, no idea how things work in Ros 2. I'd start checking whether all topics subscribed by the controller are coming at reasonable rates.

peci1, 2022-05-20:
I'm not sure how you constructed the models. But if you took Spot from the SubT Challenge tech repo, it is only tested in Ignition Gazebo Dome and with DART dynamics engine. If you run it in Gazebo Classic with ODE, the results could be anything...

jediofgever, 2022-05-20:
> I'm not sure how you constructed the models.

As for models, I did not change anything in xacros, just added the `ros2_control` tags. links, interias are all the same as the original ROS1 versions.

>  it is only tested in Ignition Gazebo Dome and with DART dynamics engine,  the results could be anything...

That is a fair point for the spot, but for other models (champ and the last robot from above comment) that I tested, they were tested in classic Gazebo.

karthikabalan, 2024-10-02:
I also have the robot spawning upside down in gazebo. This is with ROS1 noetic. Has anyone else faced this issue ?