chvmp/champ #46: The robot is not moving forward

deardost, 2021-03-23:
Hi there
I tried launching the files below:
[image]

But I ran into the following problems:
1. Gazebo environment is running in a very slow speed. (My system is Core i5, with 8 gb ram and 2 gb AMD Radeon Graphics Card. I am using Ubuntu 20.04.2 LTS and ROS Noetic. It's installed in SSD). I wonder if my system specs are not sufficient for normal speed.
2. The robot is doing somewhat rotation almost around it's axis. Would you please tell me why it happens and what's the solution?
Thanks in advance for all this help
I have attached a video below to show the warnings that I get and to show what the robot does:

https://user-images.githubusercontent.com/74898176/112160318-0c614a00-8c10-11eb-8307-024c079ac467.mp4

grassjelly, 2021-03-23:
Your RTF's way too low. Your problem might be related to this: https://github.com/chvmp/champ/issues/14

Alternatively if you just want to try walking the robot, you can try simulating without the assets:

```
roscd champ_config/worlds
roslaunch champ_config gazebo.launch gazebo_world:=default.world
```

and use teleop to control the robot

peci1, 2021-03-23:
Solution for proper running using simulation time has been just proposed: https://github.com/chvmp/champ/pull/48 .

deardost, 2021-03-26:
@peci1 Forgive me, but I don't understand your proposed solution. I am a noob in github. What do I need to do with the package (champ) that I have cloned to make it run smoothly and not slow?

deardost, 2021-03-26:
@grassjelly I want full navigation for our robot. We are working on a project. It's quadruped mobile robot. So we want our robot to navigate to the target location, using gps

peci1, 2021-03-26:
Just call `git pull --recurse-submodules` in the directory where you cloned champ. I'm not sure about Noetic compatibility, but I don't see any general problem for it to not work.

deardost, 2021-04-01:
@peci1 nope. That didn't solve my issue. Robot is still spinning around before it reaches final destination.