chvmp/champ #49: Robot keeps spinning around / colliding with walls under SLAM mode

dlee640, 2021-03-26:
Hi. I've tested champ default quadruped on willowgarage.world model. When I send 2D Nav Goal via rviz, the robot does slowly approach the destination, but keeps spinning around (seems like the robot gets confused on how to fix yaw) until it gets to that position instead of getting there in a clean fashion. Also, the robot has trouble avoiding walls, as the door passage in the willowgarage is quite narrow and the robot seems to not make it past the door.

I am attaching gifs demonstrating these issues. I noticed from previous post that PR is submitted on #48 for sim time issue and that this behavior may be related to that. However, I am lost on how to fix sim time issue. Can someone help with step by step instruction on how to address this issue?

Thanks a lot!

[image]
[image]
[image]

deardost, 2021-04-14:
@grassjelly 
@peci1
Any solution to this?? I don't know how it's working normal on some people systems. I am also facing the same issues; **Slow**, **Spinning around**, **Colliding**, **Deviating from it's path**

deardost, 2021-04-16:
@dlee640 My issue got solved. Actually I deleted back the package. While reinstalling....
[image]

IF you see 1.1 in the pic above, the first line of code is installing python-rosdep. Last time when I ran this first line of command, it threw some error. I ignored and carried on further steps. This time, I changed it.
`sudo apt install -y python-rosdep` is for earlier versions. To be able to install it on ROS-Noetic, you have to change it to `sudo apt install -y python3-rosdep`

This step worked for me and the issue of spinning of the robot did not appear this time.
Hope it helps.

grassjelly, 2021-06-10:
hey @dlee640 have you resolved this issue?