chvmp/champ #51: What's it using jfor localization?

deardost, 2021-04-17:
@grassjelly @tarasborov @peci1 @eborghi10 @SonnyPoint Doesn't amcl node works on particle filter i-e monte carlo localization?
If yes, does your package, champ, also work on that basis? The thing is that I don't see the particle filtration happening in the map. Like I don't see a visual of it where it shows particles spread throughout the map and then narrowing it down to the robot.

(Sorry if this is not place to ask the question on github. I am a noob and I don't know it yet. Please let me know where can I ask questions like these in the future, it it's not the right place)

grassjelly, 2021-04-17:
Hi, please don’t unnecessarily tag people when posting issues next time. 

CHAMP uses AMCL to localize the robot. https://github.com/chvmp/champ/blob/master/champ_config/launch/navigate.launch#L16 . It’s pre-configured on autogenerated config packages. If you want to see the particles, just add the particles cloud in RViz http://wiki.ros.org/navigation/Tutorials/Using%20rviz%20with%20the%20Navigation%20Stack

deardost, 2021-04-17:
thank you. Sorry. Surely I won'tj.

closed 2021-04-26 by grassjelly