chvmp/champ #52: How do I implement this simulation on actual hardware?

alphabeta97, 2021-04-24:
I'm working on a project and I was able to simulate my model in RVIZ (still might need some tweek to it). Now, how do I implement this simulation on actual hardware? My model is simple and I won't be using many sensors for now, so I'm planning to use Arduino Mega to run servos. I'm new to this ROS and Robotics environment.

Thank You champ team for making this repository to make simulating quadruped easy. 👍 👏

robosherpa, 2021-04-25:
@alphabeta97 ROS will need an processor that can run an operating system. You may want to hook it up to a Raspberry Pi (alike) Single Board PCs for a standalone system. For your tabletop experiment, you also have an option to directly send rosserial, handy notes [here](https://newscrewdriver.com/2020/08/05/notes-on-ros2-and-rosserial/) to publish commands from ROS running on your PC to Arduino Mega.

All the best!

grassjelly, 2021-04-26:
Hey @alphabeta97 . I wrote a hardware integration guide here: https://github.com/chvmp/champ/wiki/Hardware-Integration .