chvmp/champ #53: error in slam.launch

alphabeta97, 2021-04-29:
I'm having an error while running "`roslaunch champ_config slam.launch rviz:=true`"
[image]

[image]

Also I'm not able to launch Gazebo. It's not moving beyond preparing your world
[image]

2306raj, 2021-06-05:
I have the same issue. Was anyone able to resolve this?

PranshuTople, 2021-07-12:
@alphabeta97 @2306raj 
I have solved this issue and pushed the changes over to my fork of this repository - https://github.com/PranshuTople/champ
You can use that. I have also requested @grassjelly for a pull request on the main repo to solve this issue. 
I hope he merges it and then no one will ever get this issue again

alphabeta97, 2021-07-12:
> @alphabeta97 @2306raj
> I have solved this issue and pushed the changes over to my fork of this repository - https://github.com/PranshuTople/champ
> You can use that. I have also requested @grassjelly for a pull request on the main repo to solve this issue.
> I hope he merges it and then no one will ever get this issue again
> 
> @grassjelly you can close this issue once you accept the pull request.

Thank you Pranshu. I'll check that out. 👍

PranshuTople, 2021-07-16:
@alphabeta97 have you tried the git fork that I recommended. Did it work for you? Any feedback/suggestion/query?

alphabeta97, 2021-07-17:
> @alphabeta97 have you tried the git fork that I recommended. Did it work for you? Any feedback/suggestion/query?

Sorry to get back to you this late but I did try it out.. It is working but I'm not able to navigate the robot. i'm attaching the screenshots.
[image]
[image]
In terminal, it sure is getting new coordinates but is not able to move in the simulation.
[image]

grassjelly, 2021-07-17:
are you on Noetic?

sasilva1998, 2022-09-18:
> Sorry to get back to you this late but I did try it out.. It is working but I'm not able to navigate the robot. i'm attaching the screenshots. [image] [image] In terminal, it sure is getting new [...]

Hello, the same is happening to me, could somebody solve the issue?