chvmp/champ #55: RTF very slow running on Jetson Nano

fus3, 2021-05-16:
Hello im running champ on a Jetson Nano the problem comes when running the gazebo simulation, because the maximum RTF achieved is 0.15 or 0.16. I tried updating real_time_update to 0 but the RTF stills the same, also i tried changing the step but then the joints go crazy. When running a turtlebot3 simulation on gazebo the RTF is about ~1. Im using the jetson through vnc viewer so i was wondering if that was the problem, so i tried the jetson through hdmi but it still happening the same, the fps are about 65 so i dont think that the problem comes from the hardware, any ideas?  As you can see i also tried deleting all the props on the scene but nothing happens.

[image]

fus3, 2021-05-16:
Also i add here a question/problem that im having when creating a node for the hardware that suscribes to /joint_group_position_controller/command/ in gazebo it publishes the positions but is not publishing any position when running bringup.launch on rviz but if input harware_connected:=true to make it publish the positions then the joints go crazy. Following the hardware page on the wiki i cannot know whats happening.

grassjelly, 2021-06-10:
I haven't tried running the simulation on Jetson Nano, but the low RTF could be due to the computational constraint on your SBC. I recommend running the simulation at least on a laptop. You can check out this thread https://github.com/chvmp/champ/issues/14

Could you post the hardware problem as another thread?

closed 2021-09-17 by grassjelly