chvmp/champ #56: Gait Pattern

alphabeta97, 2021-05-24:
Which Gait Pattern does CHAMP incorporate for walking?

mvipin, 2021-05-25:
Coincidentally, I sent out an email with a similar question 2 days back. I am yet to hear from Juan. Looking at the code, I have a feeling that the following lines in the phase_generator.h might have some role in deciding the gait.
```
               leg_clocks[0] = elapsed_time_ref - (0.0f * stride_period); 
               leg_clocks[1] = elapsed_time_ref - (0.5f * stride_period); 
               leg_clocks[2] = elapsed_time_ref - (0.5f * stride_period);
               leg_clocks[3] = elapsed_time_ref - (0.0f * stride_period);
```
Most likely a trot?

alphabeta97, 2021-05-25:
> Coincidentally, I sent out an email with a similar question yesterday. I am yet to hear from Juan. Looking at the code, I have a feeling that the following lines in the phase_generator.h might have some role in deciding the gait.
> 
> ```
>                leg_clocks[0] = elapsed_time_ref - (0.0f * stride_period); 
>                leg_clocks[1] = elapsed_time_ref - (0.5f * stride_period); 
>                leg_clocks[2] = elapsed_time_ref - (0.5f * stride_period);
>                leg_clocks[3] = elapsed_time_ref - (0.0f * stride_period);
> ```
> 
> Most likely a trot?

You might be true about that... 😂 I too think It might be Trot gait coz it can be seen while we simulate the model (atleast I guess that). 
Let's wait and see if we hear. 
If you hear anything from Juan pls. do let me know too at pranav97jain@gmail.com
Thank You.. 👍

grassjelly, 2021-05-25:
You can switch gaits by changing the phase lag constants in the phase generator (https://github.com/chvmp/libchamp/blob/master/leg_controller/phase_generator.h#L98-L101). Reference for various phase lag constants here: out https://github.com/mit-biomimetics/Cheetah-Software/blob/master/common/src/Controllers/GaitScheduler.cpp#L282-L424 for a full list of phase lag constants.

closed 2021-06-10 by grassjelly