chvmp/champ #57: Error with champ robot model

mgl-sg, 2021-06-09:
Ran the following commands: 
roslaunch champ_config gazebo.launch
roslaunch champ_config navigate.launch rviz:=true

But there was some issue with the robot model as seen in the image below. Additionally, the cost maps in front of the robot shows phantom lines. Not sure why this is the case?
[image]

grassjelly, 2021-06-10:
which ROS distro are you using?

mgl-sg, 2021-06-11:
I am using ROS melodic, I did managed to solve these issues by running launching the joint_state_publisher node, not sure if there is a better way?

grassjelly, 2021-06-11:
That's strange because ros_control is supposed to publish the joint_states when in Gazebo. Are you able to walk the robot?

mgl-sg, 2021-06-12:
Yes I am able to walk the robot using the champ_teleop pkg, the specfic changes I made are as follows 
1. Lauching the joint_state_publisher node within decription.launch
2. Include the joint_state_publisher gazebo plugin for gazebo in champ.urdf
3. Lastly the key thing was setting the controller spawner in gazebo.launch to continuously respawn. As I kept getting the following warning 
[WARN]: Controller Spawner couldn't find the expected controller_manager ROS Interface

grassjelly, 2021-06-12:
I see. Do you have ros_control and gazebo_ros_control installed? Btw @mgl-sg , do you happen to be from Singapore?

mgl-sg, 2021-06-12:
Yes I have both installed, had a friend try to run the same simulation and he had the same issues as well until he made the changes above. Yes I am from Singapore

grassjelly, 2021-08-15:
cool! Is there any way I can reach out to you? Email?  @mgl-sg

closed 2021-09-17 by grassjelly