chvmp/champ #62: catkin_make error

syhassan, 2021-08-06:
Hello,

I'm using **ROS Noetic with Ubuntu 20.04** and facing an error when executing catkin_make of a missing header file, urdf_loader.h as follows:

[image]

I'm a bit of a beginner, so thank you for your help!

Best regards,
Syed

robo4869, 2021-08-10:
Did you do the installation as instructed? Here is your loss file: https://github.com/chvmp/libchamp

robo4869, 2021-08-10:
In here: https://github.com/chvmp/champ/blob/master/.travis.yml I see 2 distro kinetic and melodic, how about you add manually noetic?

grassjelly, 2021-08-15:
Hey @syhassan did you clone the repo recursively  as instructed here? https://github.com/chvmp/champ#11-clone-and-install-all-dependencies

syhassan, 2021-08-17:
Thank you for your help! It turns out I wasn't cloning it recursively as you stated!

closed 2021-08-17 by syhassan