chvmp/champ #65: How can I run robot in rest_mode?

robo4869, 2021-08-29:
Hi @grassjelly and champ developers,
I'm stuck in the control robot in rest_mode using the keyboard. I saw mini-cheetah in [this gif](https://raw.githubusercontent.com/chvmp/champ/master/docs/images/robots.gif) do it. How can I run?

grassjelly, 2021-08-29:
Hi, you would need a joystick to control the body's pose. Alternatively you can create a publisher using geometry_msgs/Pose to 'body_pose' topic.

robo4869, 2021-08-29:
What is your idea about rest mode?
I think the idea is as follows:
First, set a new pose of the body, then set a new position of each toe as the current value. Then calculate angle through inverse kinematic formula, right?

grassjelly, 2021-08-29:
you can do that by changing the z value which is the height of the robot

robo4869, 2021-08-29:
That's exactly what I've seen in other developers. I don't obvious it. Please tell me more

grassjelly, 2021-08-29:
You can use this as a starting point https://github.com/chvmp/champ_teleop/blob/master/champ_teleop.py#L102-L120 .
Rather than using the values from the joystick, you can slowly increase/decrease body_pose_lite.z until the robot reach the height that you want

robo4869, 2021-08-29:
Thanks, I also asked about whether or not possible to do rest_mode following steps which I mentioned above at r/robotics on Reddit. After that, I read your solution a few days ago and what I do not understand is the math behind it. It seems strange to me. There's nothing better than having it explained by its own author. That's why I open this issue. Can you explain it here? Or you can contact me via chienvt197@gmail.com if it needed.

robo4869, 2021-08-29:
I wanna do rest_mode before do any other gait.  So it can be as an experiment.

Quothd, 2021-08-29:
I have programmed a transition to z0.5 and repeatedly send z = 0.5. This gives me a rest pose. I also have an inactivity route be that transitions to rest after 30 seconds and turns off power to the servos after 60.

robo4869, 2021-08-29:
@Quothd You did it on a real robot? Unfortunately, I don't have a real one here. But I would like to see your, thanks

Quothd, 2021-08-29:
I'm away this weekend but realised that because mine is spending so much time on its stand, I needed more control so I modified the teleop node to support a web interface with touch button for stand / lie down, pose / walk and power control for the servo relay. 
The code needs a lot of cleaning up but happy to share if interested

robo4869, 2021-08-29:
@Quothd Have you get champ's idea about rest_mode? Please tell me! I'm stuck in few days.

grassjelly, 2021-08-29:
Closing this for now as this is not really an issue.

closed 2021-08-29 by grassjelly

robo4869, 2021-08-30:
Ok, sorry champ's author