chvmp/champ #66: champ package possible on noetic, or ros2 ?

elpimous, 2021-08-30:
Hello,
Could i install champ package on ubuntu20, ros noetic, or ros2 ?
Thanks, 
Have a nice day

elpimous, 2021-09-02:
Hey, called the py file as py3,
Some ros py3 packages, 
And Success.

lubitz99, 2021-10-19:
@elpimous Would you be able to let me know how you were able to update the code for ROS Noetic. I am struggling to make the workspace with python3.

wmmc88, 2021-10-29:
Anyone have any success running this on ROS2?

jediofgever, 2022-05-17:
Hi,
I have spent time porting a good chunk of champ code to ROS2 [here](https://github.com/jediofgever/quadruped_ros2),
There is an issue that I don't understand and I would be happy to collaborate with anyone interested running champ in ROS2. 

I suspect the issue is with controller, the robot keps wobbling in Gazebo as in below;

https://user-images.githubusercontent.com/18460336/168811789-5f4e9ae6-2734-401d-b22a-0e3851b7d3c4.mp4

sskorol, 2022-07-11:
@jediofgever have you figured out your issue?
@grassjelly do you plan to create an official port for ROS2?

grassjelly, 2022-07-17:
@sskorol you can checkout the ROS2 branch https://github.com/chvmp/champ/pull/86

closed 2023-10-14 by elpimous