chvmp/champ #67: Dear professor, I met a problem when I was runing the demo.

luckyt0204, 2021-09-18:
As the code in champ_teleop teleop.py , some keys were designed to control the pose of the robot. But when I run the demo.launch of the Chiken_head ,  there were only some keys that work normally.  They are u,i,o,j,l,m,,,. to control the velocity of the robot. F H R T Y B N are not useful. I use rostopic echo /body_pose command and discover that there are not msgs published in this topic. I wonder why I can't control  the pose of the robot. Thank you.

grassjelly, 2021-09-19:
Hey, I removed the keys to control the body's pose. You'll need a joystick for body pose control.

luckyt0204, 2021-09-28:
Thank you. But is there a requirement for the brand of the joystick? PS or xbox？

luckyt0204, 2021-09-28:
> Hey, I removed the keys to control the body's pose. You'll need a joystick for body pose control.
Thank you professor. But is there a requirement for the brand of the joystick? PS or xbox？And I want to control it to recover from falling. Would you please give me some advise?