chvmp/champ #68: Robot is just Bouncing in Gazebo

ShubhamNandi, 2021-09-24:
Hello I'm linking the [Repo](https://github.com/ShubhamNandi/dd_dot-
![dd_dot]) 
[image]
I have made this robot model with [Fusion2Urdf](https://github.com/syuntoku14/fusion2urdf)
Please let me know what have i missed!

RamCharanThota, 2022-07-26:
Even I faced the similar issue. check the PID values in control.yaml file inside the your robot config package. Especially start with small P values.

abdu7rahman, 2023-03-31:
any update on how you managed to solve this issue?

ShubhamNandi, 2023-05-03:
Yes, so the issue is with the PID. This time, I didn't change the PID and I'm using the default values. It's at least not exhibiting the same behavior, but I'm facing new issues now
Following is the error msg 
"TF_OLD_DATA ignoring data from the past for frame rf_shoulder_1 at time 38.73 according to authority unknown_publisher
"