chvmp/champ #69: Spot 2D Nav Goal not working in AWS Robomaker

JAYapeter, 2021-09-27:
Hi! I'm running a Robomaker simulation using the the Champ package with Spot. When I try to run "roslaunch spot_config navigate.launch", I get this error "[ WARN] [1632765900.092520249, 74.976000000]: Timed out waiting for transform from base_footprint to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.1 timeout was 0.1." When I try using the 2D Nav Goal function in rviz, nothing happens. It seems like it's missing transforms between the map and the body. What can I do to fix this?

JAYapeter, 2021-09-27:
In rviz, if I set the fixed frame to "body", I get an error saying there is no transform from map to body. 
[image]

And if I set the fixed frame to "map", I get errors saying there is no transform from each part of the body to map, and LaserScan also throws an error.
[image]

grassjelly, 2021-09-28:
do you have gmapping installed/running in the background?? https://github.com/chvmp/champ#11-clone-and-install-all-dependencies

JAYapeter, 2021-09-28:
I used to command "sudo apt-get install ros-melodic-openslam-gmapping" to double check that I do have gmapping installed. I still have the same issues when I run my simulation. How can I double check to make sure gmapping is running when my simulation is running? Thanks!