chvmp/champ #75: Change step height

itaouil, 2021-11-22:
Hi @grassjelly,

Thanks a lot for making this open source, and for actively working on it :).

I wanted to create a new gait or modify one of the existing ones to get AlienGo to climb stairs blindly on simulation.

I saw issue #56 and I was wondering whether there is a parameter that I could change to raise the step height? And of course if you have any insight or suggestions regarding being able to blindly climb stairs with AlienGo using champ :)?

Best,
Ilyass

grassjelly, 2021-12-07:
Hey @itaouil This project might be useful. https://github.com/hgaren/aliengo_delivery/blob/master/aliengo_base/src/quadruped_awareness_controller.cpp#L166_delivery. They altered the gait pattern and use https://github.com/ANYbotics/grid_map to estimate a good foot location to climb stairs.

Supergithubber, 2021-12-12:
Hi！
@itaouil @grassjelly 
After configuring Aliengo with Assistant, aliengo can walk in Rviz, but cannot stand in Gazebo. How can I solve this problem?
Thank you for your answer！

itaouil, 2021-12-12:
> Hey @itaouil This project might be useful. https://github.com/hgaren/aliengo_delivery/blob/master/aliengo_base/src/quadruped_awareness_controller.cpp#L166_delivery. They altered the gait pattern and use https://github.com/ANYbotics/grid_map to estimate a good foot location to climb stairs.

Thank you @grassjelly

itaouil, 2021-12-12:
Hi @Supergithubber,

I am not sure what you are referring to.

I did not try what Juan suggested yet, if that is what you mean by "Assistant" :).

closed 2021-12-12 by itaouil

Supergithubber, 2021-12-12:
> 
Hi，Thank you for your reply！
I encountered the following  problems:
When I used Aliengo, it walked in Rviz, but in Gazebo it lay prone and couldn't walk.
Looking forward to your reply!

amit-z, 2022-11-03:
@itaouil Hey, did you manage to modify Champ so that it can climb stairs? It seems that the script we should modify is [quadruped_controller](https://github.com/chvmp/champ/blob/master/champ_base/src/quadruped_controller.cpp). 
I'll be happy to hear if you have some feedback regarding this topic.