chvmp/champ #76: Odometry

xxd-xy, 2021-12-07:
How the odometry of the quadruped robot was realized in this project ?

grassjelly, 2021-12-07:
You can take a look at the implementation here: https://github.com/chvmp/libchamp/blob/master/odometry/odometry.h

chrisalbertson, 2021-12-07:
I read the above link.    I like to do it differently.  Just use the
commanded motion of the body.  Basically dead reckoning.

The linked code assumes the robot has zero velocity if (1) all feet are in
ground contact.   No.  The robot might be in the process of leaning in some
direction and hence moving.  I define "moving" as the body center point is
translating relative to the real world.  Leaning is a valid way to make
small movements.  (Perhaps to better-position a camera?) and (2) the linked
code assumes if all feet of not in contact the robot is stationary.    No.
It might be an actual MIT Cheetah that is powerful enough to use a bound or
pronk gait.   (The linked code will always return "zero" in the case of
pronking.)  Yes, any robot I could personally afford is upside down if all
feet are not in ground contact.  But I can hope this situation changes.

Dead Reckoning is not accurate over long distances so I am looking at
visual odometry, perhaps a sensor that faces down to the floor.  Basically
an optical mouse

grassjelly, 2021-12-09:
You could perhaps implement your own odometery implementation (something like this https://github.com/chvmp/champ/blob/master/champ_base/src/state_estimation.cpp#L110-L165) and fuse it with the other odometry/sensor source in either one of the ekf configs here: https://github.com/chvmp/champ/tree/master/champ_base/config/ekf

closed 2022-07-02 by grassjelly