chvmp/champ #82: where is gmapping code?

juno100, 2022-04-18:
i want to study and change the gampping code

where is gmapping code in champ?

chrisalbertson, 2022-04-18:
Wow, I know this one...

The actual code is in the ROS GitHub in the gmapping package but the stuff
you want to possibly change is in the launch file at
https://github.com/chvmp/champ/blob/master/champ_navigation/launch/slam/gmapping.launch

Documented here http://docs.ros.org/en/hydro/api/gmapping/html/index.html

juno100, 2022-04-18:
can i change the code?

grassjelly, 2022-04-20:
Yes you can, but sorry that's not part of this project's scope. Closing for now. Thanks

closed 2022-04-20 by grassjelly

chrisalbertson, 2022-10-11:
I tracked down the gmapping library that ROS uses.  It seems to be about 15
years old and has not been modified in 4 years. Then I found the fork that
is used by ROS.   It has recent changes and seems to be what an actual
robot would use today
https://github.com/ros-perception/openslam_gmapping

If you are going to modify this, I'd set up a simple test robot in gazebo,
use Turtlebot.