chvmp/champ #84: Pointcloud2 not working in simulator

amirandamoreno, 2022-05-25:
Hello all,

I hope you can help me to solve this issue, I was trying to work with the unitree cloning the model config as indicated,
the problem comes when i run the following launching file:

roslaunch go1_config gazebo.launch 

the simulator works as expected, but the problem comes when I try to visualize in rviz the pointcloud2 /camX/point_cloud_XXXX.

It gives an error saying: "Failed to transform from frame [/camera_optical_face] to frame [xxxxxx]. I this that / at the beginning of camera_optical_face is making things work wrong. Is there any idea on how to solve this issue? 

I'm sending the picture of rviz so that you can understand better the problem

[image]

grassjelly, 2022-07-02:
Hey @amirandamoreno . I believe this is related to the plugin used in Go1's URDF https://github.com/unitreerobotics/unitree_ros . It's not supported in this repository unfortunately. However you could simulate the poincloud by adding a standard sensor and map it to the same sensor topic.

grassjelly, 2022-07-02:
Closing this for now.

closed 2022-07-02 by grassjelly