chvmp/champ #89: Compile my controller under champ_base

elpimous, 2022-06-29:
Hello,
I made my own lib (working and tested)
Now, I inserted my header and it cpp on include and src of champ_base,
Modified it cmakelist.txt,
And I create one instance of my lib class, into quadruped_controller.cpp.
When catkin_make,  I encounter compilation error..
Any idea? It must be simple, but can't find the solution.

https://github.com/elpimous/ylo-2/issues/1

Thanks a lot

grassjelly, 2022-07-02:
hey I looked at the code: https://github.com/elpimous/ylo-2/blob/d4f7856ec04cfa64a67294d47243ecbd0d0898a3/champ_for_ylo2/champ_base/src/quadruped_controller.cpp#L30 . Are you supposed to instantiate that object inside the constructor? Anyway closing this as the error is not related to champ.

closed 2022-07-02 by grassjelly

elpimous, 2022-07-02:
It seems I had success with compilation.. 
Non, need to find where to put my instantiated object.. 
Thanks friend