chvmp/champ #91: Champ on real robot.

elpimous, 2022-07-03:
Hello, 
Ylo2 real robot works on Champ (BLDC motors, working on Moteus controllers, using position mode), but anyway, a few questions : 
1- Standup : Need to make a standup function : where should i call it, in your code ?
2- Robot height, when walking : where to change Robot Walking Height ?
3- Sitdown : where should i call it, in your code ?

elpimous, 2022-07-03:
2 robot_height can be changed here :  <your_robot_name>/config/gait.yaml/nominal_height : 0.3

grassjelly, 2022-07-03:
standing up and standing down can be done by changing  the values of the target foot position sent to the inverse kinematics engine. This line changes those values https://github.com/chvmp/champ/blob/master/champ_base/src/quadruped_controller.cpp#L98 depending on the pose defined by the user. In your case you can get the initial foot position using the joints and perform forward kinematics and move towards a foot position that will make the robot stand / sit down over specified period of time.

elpimous, 2023-03-21:
👍