chvmp/champ #93: Make Gazebo controller respawn option configurable

sskorol, 2022-08-15:
Currently, Gazebo `controller_spawner` has a [hardcoded](https://github.com/chvmp/champ/blob/master/champ_gazebo/launch/gazebo.launch#L31) `respawn="false"` arg. There are no issues while running it locally. However, if you try launching `gazebo.launch` in a Docker, `controller_spawner` just times out and exists. It blocks any further teleop attempts as joints' data is not managed anymore. The issue can be fixed by switching `respawn="true"`. In this case, after several timeouts, when Gazebo is ready, this node can be successfully spawned. I propose to make this option configurable so that anyone can pass it as an argument.

grassjelly, 2022-08-15:
could you send a PR?

sskorol, 2022-08-15:
@grassjelly just sent :)

PR #94 (2022-08-15): GH-93: make controller_spawner respawn configurable
Added an additional Gazebo arg to be able to automatically respawn controller_spawner node.

Fixed #93
closed 2022-08-15 by grassjelly