chvmp/champ #101: Robot legs move in opposite directions than what's showing on rviz using joints gui

TonyLiu836, 2022-09-04:
Hi,

I've been stuck on trying to figure out why the legs on the left side of my robot move in opposite directions than what's showing on rviz when using the joint gui. I followed the hardwar_integration guide and mounted the servos at their zero positions with the robot's legs pointing straight down. Additionally, the shoulder/hip actuator can only move in one direction, as it is already at the limit of it motion range according to the directions in the hardware_integration guide. Is this intended? 

Thanks!

RamCharanThota, 2022-09-08:
On of the Simple check we did is, whether all the axis defined in the leg joints follow the required champ configuration which is as shown in figure
[image].
Hope it will help to resolve the issue

closed 2022-09-27 by TonyLiu836

yoggi56, 2023-08-17:
I encountered the same issue when I had wrong rotation directions in joints in URDF file of my robot. Check champ.urdf file. You should have these values for your robot:
1. hip_joint: <axis xyz="1 0 0"/>
2. upper_leg_joint: <axis xyz="0 1 0"/>
3. lower_leg_joint: <axis xyz="0 1 0"/>