chvmp/champ #102: Leg Odometry

RamCharanThota, 2022-09-08:
Hi, 
I am working on quadruped robot project. I have successfully drive the robot around (only couple of meters) using champ package. But, I found out some issues in robot position estimated by robot localisation package. So, I was trying to understand how you are doing leg odometer. But, I couldn't wrap my head around it. I am wondering is there any academic article or paper which would help me to understand logic behind champ leg odometer. I really appreciate if you can provide me with such information.

grassjelly, 2022-09-25:
First, the robot calculates each leg's position (ref to inertial frame)by doing forward kinematics and then differentiate those positions to get the velocity

RamCharanThota, 2022-09-26:
Thank you Juan for reply and answering my question

closed 2022-10-28 by grassjelly