chvmp/champ #105: Maybe some TF transform problem

HB-Liang, 2022-10-26:
Hi, recently I started using this program and I met some problem
I launch the demo using
"roslaunch champ_config gazebo.launch robot_name:=quadruped world_init_x:=1.0 world_init_y:=1.0"
to test some origin position changing & prepare for multi-robot system.
But when I launch the navigation program, with
"roslaunch champ_config navigate.launch robot_name:=quadruped"
something wrong in rviz. The laser scan is not aligned with the map.
At the very start on rviz launch, I notice that the frame "quadruped/odom" is not pretty well at the position (1.0, 1.0)
where should I start my diagnose process?
Thanks. 
[image]

grassjelly, 2022-10-27:
Hi, you'll need to reinitialize the robot's pose using RVIZ to align the LIDAR.  Since you started the robot at (1,1), the robot has no idea that it's at location reference to the origin (where the robot started when the map was generated).

HB-Liang, 2022-10-28:
Thanks! The robot doesn't know where it is, that makes sense. 
Did you mean that if I want to do more application with starting position at (1,1), I need to generate another map with that starting position? 
Further, if I need two or more robot working in the same environment, they have to have a map with their own starting position, or some multi-robot slam technique.

grassjelly, 2022-10-28:
you just have to programmatically tell the robot where it is. You can do that by publishing to initialpose topic of the robot’s known initial pose which in this case (1,1)

HB-Liang, 2022-10-28:
This helps a lot, Thank you!

closed 2022-10-28 by HB-Liang