chvmp/champ #106: way to control the force with which robot foot stepping on the ground

RamCharanThota, 2022-11-11:
Hi Miguel,
I am implementing the champ on the real robot. But, I have observed very high ground reaction force on feet whenever foot is in contact. I am wondering is there way to control the force with which robot foot stepping on the ground. Thank you and l really appreciate it if you can answer my question.

grassjelly, 2022-11-11:
Hey @RamCharanThota . That's the current limitation of the default position controller. One way you can limit these GRFs is to do a compliant controller on the joint level.

RamCharanThota, 2022-11-14:
Thanks @grassjelly  for answering my question.  I really appreciate it.

RamCharanThota, 2022-11-18:
Hi Miguel (@grassjelly),
Sorry to bother you again. I have read some papers on leg controllers.I come across impedance control implemented (which is mentioned in MIT cheetah paper suggested by you). I don't have much experience in the controls. I am wondering whether I can use this impedance control to control the ground reaction forces. If not , can you suggest any journal or resource which can help me to limit these GRFs. Thank you and Looking forward to your reply.
]

grassjelly, 2022-11-29:
hey @RamCharanThota this might be a good start

grassjelly, 2022-11-29:
closing this for now

closed 2022-11-29 by grassjelly

RamCharanThota, 2022-11-29:
thank you Miguel (@grassjelly )